OpenBCI/Documentation #6: move several docs out of Deprecated

wjcroft, 2019-09-08:
Three documents should be moved out of deprecated: the two docs on FTDI adjustment; and the doc on VirtualBox.

A number of VirtualBox users of BrainBay, BioEra, or OpenViBE, depend on VirtualBox to run Windows apps on their Mac or Linux system. I fully understand that OpenBCI users should NOT run the GUI via VirtualBox. This Windows apps workaround is only for official Windows only apps.

Regarding the FTDI docs. These should be titled something like: "FTDI buffering adjustment on Mac" and "FTDI buffering adjustment on Windows".  Along with this change, it may be helpful to do a benchmark test on a totally new Mac or Windows install, to see if these docs still apply to the latest versions of those respective OSes. They certainly HAVE been necessary for past versions of the OSes, and thus should not be deprecated. One could add a disclaimer to the top of the docs, stating which OSes may no longer need the fixes.

Regards, William

retiutut, 2019-12-16:
Probably still applies to current OS. I can test this on Mac and Windows. The docs have been undeprecated!

wjcroft, 2019-12-16:
I'm still seeing the VirtualBox and FTDI Mac docs in the Deprecated section. And ZERO trace of where the Windows FTDI doc went. Suggest this issue is not yet 'Closed'.

wjcroft, 2019-12-16:
OK, see the FTDI docs now in Troubleshooting. But FTDI Mac doc still lists in deprecated.

I thought VirtualBox doc was to be moved to Software.

retiutut, 2019-12-16:
For now they are all in troubleshooting. Might be because of cache.

wjcroft, 2019-12-16:
Might be a caching issue. VirtualBox doc should definitely go into Software, under Third Party, along with BioEra, BrainBay and OpenViBE, which are all Windows native. So VirtualBox is only way to run these on Mac.

retiutut, 2019-12-16:
Still publishing, but I can make that change.

closed 2019-12-16 by retiutut