OpenBCI/Documentation #13: Erroneous descriptions in Documentation?

Otemo, 2019-10-20:
I have found some confusing descriptions in the “Documentation” OpenBCI recently renewed.

1 [Setting up for EEG ](https://docs.openbci.com/docs/01GettingStarted/02-Biosensing-Setups/EEGSetup)

In the table describing how to connect the electrodes to the Gannglion board, there is a description I cannot understand.
There are no "SRB" and "Bias" pins on the Ganglion board.
Here is my idea !

[image]

2 [Ganglion Specs ](https://docs.openbci.com/docs/03Ganglion/GanglionSpecs)

The documentations in the website tells us to connect the ground electrode to **Both** the top **And** bottom of the D_G pin. The similar explanations are given for the REF pin.
However in another web pages, they tell me i need to connect **One** electrode to the top D_G (REF) pin.

・[Headband-Ganglion Tutorial ](https://docs.openbci.com/docs/04AddOns/01-Headwear/HeadBand#headband-ganglion-tutorial)

Does it mean i may connect to the top **OR** bottom of D_G (REF) pin?
I suppose it is enough to connect the ground(reference) electrodes to the top **OR** bottom pins.

I think i have been badly mistaken... ??
I would appreciate knowing more about it.

Otemo, 2019-10-30:
The support team solved the # 1 problem. 
Thank you so much. 

・EEG setup with Ganglion:
https://docs.openbci.com/docs/01GettingStarted/02-Biosensing-Setups/EEGSetup#1-connect-your-electrodes-to-openbci

・EMG setup with Ganglion:
https://docs.openbci.com/docs/01GettingStarted/02-Biosensing-Setups/EMGSetup#1-connecting-electrodes-to-the-openbci-board

”[Ganglion Specs](https://docs.openbci.com/docs/03Ganglion/GanglionSpecs)” has been still an open question for me.

retiutut, 2019-10-30:
@Otemo Can this issue be closed?

Otemo, 2019-10-30:
Hi @retiutut,

Thanks to your support, issue # 1 has been solved.
However, issue # 2(”[Ganglion Specs](https://docs.openbci.com/docs/03Ganglion/GanglionSpecs)”) is still unresolved. 

If you think i have been badly mistaken, please let me know.

retiutut, 2019-12-13:
From @wjcroft:

Both the schematic and layer diagrams, show that top and bottom pins of D_G and REF are connected together with traces. So makes no difference which you use.
https://docs.openbci.com/docs/assets/GanglionImages/ganglion_schematic.png

Otemo, 2019-12-13:
Hi @retiutut ,
Thank you for your very polite explanation!
Please close this issue.

@Otemo

retiutut, 2019-12-13:
About to add this information to the Docs!

closed 2019-12-13 by retiutut