OpenBCI/Documentation #16: "...adjust your channel settings" section missing in new docs

wjcroft, 2019-12-11:
The old Cyton Getting Started guide, had a section on how to adjust the settings panel in the GUI. I cannot seem to find this anywhere in the new docs. Here is the old docs link:

https://github.com/OpenBCI/Docs/blob/master/Tutorials/01-Cyton_Getting%20Started_Guide.md#4-launch-the-gui-and-adjust-your-channel-settings

In the case mentioned above, the user is seeing EEG noise when doing simultaneous EEG, EMG, ECG. The solution being to remove the stronger signals from the common reference bus, and change the scale factor, etc.

Admittedly not a common case. But some EMG, ECG users do have need for customization of these settings.

They ARE mentioned in the SDK doc. But that is geared to programmers, not GUI users.

retiutut, 2020-06-09:
#25 Requested from another user

wjcroft, 2020-06-30:
Requested again by user BioPotential,

https://openbci.com/forum/index.php?p=/discussion/comment/13853/#Comment_13853
https://openbci.com/forum/index.php?p=/discussion/2599/cyton-v4

I pointed them at the link above to the old doc. Question, how does that link automatically redirect to the OpenBCI-archive tree?? Github wizardry!

https://github.com/openbci-archive/Docs/blob/master/Tutorials/01-Cyton_Getting%20Started_Guide.md#4-launch-the-gui-and-adjust-your-channel-settings

wjcroft, 2021-04-28:
Eva, hi.

As the original post on this issue mentioned, this old doc section showed how to adjust the Cyton channel settings,

https://github.com/openbci-archive/Docs/blob/master/Tutorials/01-Cyton_Getting%20Started_Guide.md#4-launch-the-gui-and-adjust-your-channel-settings

[scroll down until you see "4. Launch the GUI and adjust your channel settings". Then the 3rd screen shown in that section is the channel adjust illustration.

I'm looking now at the current docs:

https://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIDocs
https://docs.openbci.com/docs/01GettingStarted/01-Boards/CytonGS

And it's not clear if there is a demo of adjusting the channel settings. Can you point me where the settings instructions are, at this current time?  Mentioning @retiutut 

Regards, William

retiutut, 2021-04-28:
@wjcroft https://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIWidgets#hardware-settings

But we agreed in a meeting today to make a new doc specifically focused on hybrid data collection: https://github.com/OpenBCI/Documentation/issues/64

wjcroft, 2021-04-28:
Richard, thanks. 

 did not realize there was a new widget section. Might be helpful to put a link to that in the current Cyton Getting Started guide. Then update to the hybrid link when that becomes available. 

William

retiutut, 2021-04-28:
- [ ] Add more information to the GUI widget guide under hardware settings
- [ ] Add link to GUI widget guide HW settings section in Cyton Getting Started guide

wjcroft, 2021-04-28:
Incidentally, although the Hardware Settings section is present in the document, it does NOT show up on the contents panel on the right. Which is why I missed it.

https://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIWidgets#hardware-settings

evaesteban, 2021-05-04:
I just added it to the right and made a PR, thanks for spotting it!

commit 3e43e721 (2021-05-04): More info on Cyton Hardware Settings - Fixes #16
closed 2021-05-04 by evaesteban