OpenBCI/Documentation #37: Mark IV Document Cleanup - Oct 2020

retiutut, 2020-10-09:
https://docs.openbci.com/docs/04AddOns/01-Headwear/MarkIV#completed-8-channel-mark-iv-setup

This doc has missing images, and the following section needs an actual chart, not bulleted points.

>For the remainder of this tutorial, the blue nodes on the 10-20 system diagram (channels 1-8 of the OpenBCI default settings) will be used. The channel to 10-20 system correlations are as follows:
Channel 1(N1P) - Fp1
Channel 2(N2P) - Fp2
Channel 3(N3P) - C3
Channel 4(N4P) - C4
Channel 5(N5P) - P7
Channel 6(N6P) - P8
Channel 7(N7P) - O1
Channel 8(N8P) - O2

Additional clarifications and cleanup as needed.

retiutut, 2020-10-09:
[image]

retiutut, 2020-10-09:
[image]

evaesteban, 2021-04-28:
Updated images and table

closed 2021-04-28 by evaesteban