OpenBCI/Documentation #39: Documentation Errors for Neuromore

jjmcgra, 2020-11-13:
Three Challenges with Documentation:

https://docs.openbci.com/docs/06Software/02-CompatibleThirdPartySoftware/Neuromore
1) Nothing specifies this information is unique to the Cyton Board (add simple note at top of documentation)
2) using the command: ls /dev/tty.* as instructed, Ganglion board is never detected as in example given

https://docs.openbci.com/docs/10Troubleshooting/FTDI_Fix_Mac
3) The FTDI Driver 2.4.2 that is recommended no longer exists on the FTDI page (Mac 10.14 - now has Driver 2.4.4)

retiutut, 2021-11-06:
This issue can be revisited now that Neuromore might be using BrainFlow.

retiutut, 2023-09-20:
Very stale issue. Reassess during an OpenBCI Software Team meeting.

retiutut, 2025-02-07:
Reviewing the Neuromore application is a great opportunity for internal use cases.