OpenBCI/Documentation #56: Add Documentation for GUI CSV Output and Header

retiutut, 2021-02-22:
https://openbci.com/forum/index.php?p=/discussion/2735/understanding-the-csv-headers-of-the-new-v5-gui#latest

This is an easy fix. Should be done ASAP.

wjcroft, 2022-03-30:
https://openbci.com/forum/index.php?p=/discussion/3288/plotting-analog-read-data-from-gui-recording-file

User confusion would be aided by more documentation. Perhaps the header could include a link to the new doc.

wjcroft, 2022-11-01:
Forum users still asking for document on CSV format:

https://openbci.com/forum/index.php?p=/discussion/2735/understanding-the-csv-headers-of-the-new-v5-gui#latest

retiutut, 2023-09-01:
[OpenBCI-RAW-2023-09-01_09-34-21.txt](https://github.com/OpenBCI/Documentation/files/12499079/OpenBCI-RAW-2023-09-01_09-34-21.txt)

retiutut, 2023-09-01:
https://github.com/OpenBCI/OpenBCI_GUI/pull/1182

PR #189 (2023-09-01): Update Cyton Aux Channel header information in GUI doc
Fixes #56
closed 2023-09-01 by retiutut