OpenBCI/Documentation #63: Electrode cap user, confused that no channel assignments are recommended (for GUI use)

wjcroft, 2021-04-01:
'Alexey' on NeuroTechX wrote this regarding his confusion over the electrode cap doc:

> ...However I have a lot of questions that the tutorials don’t seem to cover. Including the order of the electrodes and verification that I am connecting the electrodes in the right location. That information seems to be missing all together. 

I wrote him a PM in response:

----

This is the default head map used in the GUI for the 16 channel Cyton:

https://openbci.com/forum/index.php?p=/discussion/1087/connecting-mark-iv-to-cyton-including-daisy-board/p1

I’m surprised this was not mentioned in the cap tutorial. It is also shown in the Ultracortex tutorial.

https://docs.openbci.com/docs/04AddOns/01-Headwear/MarkIV#electrode-location-overview

In your own case, you can assign any scalp location to any channel. (edited)

----

So some links similar to this should be inserted in the electrode cap tutorial doc.

Regards, William

retiutut, 2021-08-19:
Another request from the Forum to clarify this information:
https://openbci.com/forum/index.php?p=/discussion/3092/mapping-for-connecting-electrode-cap-positions-to-channels

retiutut, 2021-11-06:
@carillons This might be an easy issue to resolve after working on the new Gelfree Electrode Cap doc.

PR #159 (2023-05-15): Add CytonDaisy default head map to electrode cap docs and description of exported Cyton/Ganglion csv files
Closes #124 
Closes #63
closed 2023-05-15 by retiutut