OpenBCI/Documentation #75: Add info on new Focus Widget

retiutut, 2021-06-24:
https://github.com/OpenBCI/OpenBCI_GUI/issues/924

Update GUI Widget Guide to include new info on updated Focus widget which uses BrainFlow metrics.

Also, update the Arduino instructrable  https://www.instructables.com/Send-Focus-Data-From-OpenBCI-GUI-to-Arduino/

retiutut, 2021-07-13:
Also, we should update the following official Doc:

https://docs.openbci.com/docs/07Examples/19-EEGProjects/FocusArduino

retiutut, 2021-09-19:
GUI Widget guide has been updated!

wjcroft, 2022-12-19:
At least one user,

https://openbci.com/forum/index.php?p=/discussion/3491/sending-focus-data-from-gui-to-arduino-for-new-focus-widget

is hoping that the FocusArduino document will be eventually updated to reflect the new widget,

https://docs.openbci.com/Examples/EEGProjects/FocusArduino/

retiutut, 2023-01-06:
#137