OpenBCI/Documentation #91: Clarify External Triggers for Cyton

retiutut, 2021-12-17:
The Cyton External Trigger doc needs to be cleaned up to reflect what actually works and has been learned. https://docs.openbci.com/Cyton/CytonExternal/

Optoisolation and using a photodiode pressed against the screen are two functional methods.

We also have an experiment that uses a photodiode (involving visual stimuli of puppies and kittens).
https://docs.openbci.com/Examples/VideoExperiment/

retiutut, 2021-12-19:
@wjcroft Shared new title idea for Video Experiment to “P300 Oddball tutorial, using external trigger”.

philippitts, 2023-06-30:
Let's track updating the external triggers documentation on [this new issue](https://github.com/OpenBCI/Documentation/issues/177), and have this one focus on updating the photodiode example.

retiutut, 2023-09-20:
I think we can leave the Video Experiment doc as it is for now. The idea here was to point to the use case and lessons learned in the Video Experiment project.

Marking this as a duplicate and moving info to the updated ticket.

closed 2023-09-20 by retiutut