OpenBCI/Documentation #100: Update Mac FTDI Buffer Fix doc in Troubleshooting

retiutut, 2022-03-08:
Mention that it may not be possible at this time to apply the fix for Apple Silicon M1+.

>Note that the FTDI “Latency Fix” tweak, is STILL not available for M1 (and eventually M2) Macs. This has already greatly frustrated an M1 customer. Who requires minimum Cyton latency for his application.

Applies to any Mac user of macOS 11.x. Both M1 and Intel.

wjcroft, 2022-03-08:
Actual working customer 'solution' will require getting an Apple developer license / ID, and re-building the FTDI driver with appropriate Info.plist settings. FTDI has REFUSED to help us, after MANY pleas via email and their forum. And not just OpenBCI customers.

https://www.ftdicommunity.com/index.php?topic=547.new;topicseen#new

This situation applies to ALL macOS 11.x customers, both M1/M2 and Intel.

It should not be too difficult to get the developer ID and build a custom driver. Could also be done for the macOS 10.x users. Can this be added to the software 'todo' list?

retiutut, 2022-12-13:
Lines of code have been added to BrainFlow which should attempt to automatically adjust the latency. This will be included in the next version of the GUI (5.1.1 or 5.2.0). We still have the Smoothing feature in the GUI.

closed 2022-12-13 by retiutut