OpenBCI/Documentation #139: Examples / EEG: Motor Imagery and MindAffect

wjcroft, 2023-01-17:
There is a long thread on the Forum with users hitting problems with the tutorial:

https://docs.openbci.com/Examples/EEGProjects/MotorImagery/

https://openbci.com/forum/index.php?p=/discussion/3202/neuropype-motor-imagery-tutorial-questions

The tutorial author refuses to respond to the questions, even when contacted at his email addresses and with a LinkedIn message. After the most recent contacts I did see from the Forum log that he had seen the latest posts, but made no comments. I believe a disclaimer should be posted at the top of this tutorial, warning that any problems will not get a response.

----

On your main page:

https://docs.openbci.com/Examples/ExamplesLanding/

You should definitely link to the MindAffect project, as it is a superior BCI paradigm to both MI and SSVEP. There is a long Forum thread with videos and links.

https://openbci.com/forum/index.php?p=/discussion/2783/mindaffect-announces-open-source-release-of-their-cvep-bci-speller-games-etc/

and

https://openbci.com/community/mindaffect-releases-open-source-bci-software/

retiutut, 2023-01-18:
This is probably the most difficult tutorial to support since it uses Neuropype, which isn't as accessible as other examples in our Docs. We may need to remove it if we are unable to support it, and then think about making a new one using alternate approach.

wjcroft, 2023-02-12:
At the minimum, consider placing an explicit disclaimer at the top of the Motor Imagery tutorial, saying that the tutorial author is not available for questions and that past users of the tutorial have encountered stumbling blocks.

wjcroft, 2023-03-28:
At this Forum thread, user Januz comments:

https://openbci.com/forum/index.php?p=/discussion/comment/19103/#Comment_19103

> Update: The folks at Neuropype told me to use the Select Range node to eliminate any unused channels that might interfere with the CSP ML model and it worked.

I'm going to ask Januz if he can suggest the best changes to clarify this to the current documentation.

Omiokane, 2023-04-05:
Greetings, it's me Januz and I encountered this specific error:

`graph loading graph structure`
`graph saving graph structure`
`graph loading graph structure`
`graph saving graph structure`
`execution started`
`execution resuming`
`Resolving data stream (name='obci_eeg1')`
`Importing XDF file Jan132023CSPTraining1LeftRightArm.xdf`
`Imported file Jan132023CSPTraining1LeftRightArm.xdf.`
`Imported file Jan132023CSPTraining1LeftRightArm.xdf.`
`Applying FIR filter...`
`CSP model is training now...`
`The following error occurred in node Common Spatial Patterns: Could not solve CSP problem. Your data might be rank deficient (e.g., not enough trials or electrically bridged channels). Try the FilterBankCommonSpatialPattern node with non-zero shrinkage.`
`LSL: Data stream obci_eeg1 acquired.`
`No non-empty chunks with only_signals set to False found!`
`CSP must be trained before it can be used`
`CSP must be trained before it can be used`
`CSP must be trained before it can be used`
`CSP must be trained before it can be used` (statement loops)

After asking on Neuropype's forums, they suggested that I use the Inspect Packet node so it would be easier for me to troubleshoot the problem. In my case, they saw that the "channel data is the key" since they noticed that half of the 16 channels are bridged (I only have 4 channels each working for my Cyton and Daisy board).

[image]

Thus, they told me to remove those 8 channels that are unused. If I do not remove them, then most probably I'm feeding those into the CSP ML model, and the data would be rank deficient with multiple channels all having the same data (all zeroes).  The Inspect packet node with the following pipeline: Import File -> Assign Targets -> Segmentation -> Inspect Packet, helped a lot since I got a good glimpse of the markers and segmentations to check if I have sound results. Due to this, the following error:

`The following error occurred in node Common Spatial Patterns: Could not solve CSP problem. Your data might be rank deficient (e.g., not enough trials or electrically bridged channels).`

made me rule out that the warning about rank deficiency is not a trial count issue, but rather the other case of having potentially "bridged channels", or specifically having channels that are disconnected. In the end, removing those unused channels using the Select Range node helped me fix the issue. I know that this isn't the only case/scenario regarding the error but I hope it helps.

wjcroft, 2023-06-26:
Please do not forget that this issue also requests that MindAffect (cVEP) be featured, as it is much superior to either MI or SSVEP.  It is also free and has abundant support / help docs.

philippitts, 2023-06-30:
Let's split this into two issues. This one should remain focused on the resolving problems with the current motor imagery example and #178 can focus on potentially adding a MindAffect example.

retiutut, 2023-09-20:
This issue is a very heavy lift. Should be discussed later during an OpenBCI Software Team meeting.

wjcroft, 2023-09-20:
> This issue is a very heavy lift. Should be discussed later during an OpenBCI Software Team meeting.

Wait, all that is being asked for is a **disclaimer** at the top of the current NeuroPype based tutorial page that this is unsupported and previous users have discovered issues. The tutorial author **refuses** to cooperate by answering issues.

Mentioning @philippitts

retiutut, 2023-09-20:
That seems reasonable @wjcroft. Though at that point it should probably be moved to Deprecated folder.

wjcroft, 2023-09-20:
Looking here, the links have already been removed.

https://docs.openbci.com/Examples/ExamplesLanding/

PR #246 (2025-03-28): Deprecate NeuroPype Motor Imagery Doc
Resolves #139
closed 2025-03-28 by retiutut