OpenBCI/Documentation #144: Fo

DashBarkHuss, 2023-01-28:
In the focus tutorial [focus tutorial](https://github.com/OpenBCI/Documentation/blob/master/OpenBCI_files/OpenBCI_GUI-FocusArduino/OpenBCI_GUI-FocusArduino.ino):

```
if (s.equals("false")) 
```
It doesn't make sense to me. S is a string made from char array `receivedChars` with 0's and 1's. For example, all of these are possible values for `s`: `00001`, `000`, `1111`, `1`. 

```
char receivedChars[numChars]; 
...
String s = receivedChars;
```

So when would s equal the string "false"?

DashBarkHuss, 2023-01-28:
This worked for me instead:
```
// not relaxed state
// if there are no 1's and s is not an empty string
if (s.indexOf('1') == -1 && s.length())

...

// relaxed state
// if there are no 0's and s is not an empty string
if (s.indexOf('0') == -1 && s.length())
```

retiutut, 2023-01-28:
https://github.com/OpenBCI/OpenBCI_GUI/blob/master/Networking-Test-Kit/Serial/Arduino_FocusFan_Serial/Arduino_FocusFan_Serial.ino

This Arduino sketch is what was created to work with GUI v5.

retiutut, 2023-01-30:
Thanks for catching this. All of the Arduino files have now been updated to check for 1 or 0 instead of true or false.

commit 4bdfd677 (2023-01-30): Update Arduino Focus firmware files

Fixes #144
PR #145 (2023-01-30): Update Arduino Focus firmware files
Fixes #144
closed 2023-01-30 by retiutut