OpenBCI/Documentation #176: Create video tutorial for Neurofly

philippitts, 2023-06-28:
Create a video tutorial that demonstrates how to hook up EMG electrodes using a Cyton and control the EMG Joystick in the GUI using them. The tutorial should also show how to stream the data via UDP and LSL. Finally, the demonstration should show how the data can be consumed in the drone-gcs to fly a drone.

No need to go into detail on how the GCS was created, but a system overview should be covered (what applications are running, how everything is connected, etc). When questions of scope come up, note that this is a usage tutorial, not a coding tutorial.

retiutut, 2023-09-20:
Reviewing the new video tutorial tomorrow! @isaacplotkin

retiutut, 2023-09-20:
Related to https://github.com/OpenBCI/Documentation/issues/175

philippitts, 2023-10-16:
Embed the video in the documentation and also move the [Neurofly section](https://docs.openbci.com/Software/OpenBCISoftware/NeuroFly_Toolkit/) to the [examples section](https://docs.openbci.com/Examples/ExamplesLanding/) of the documentation.