OpenBCI/Documentation #178: Add MindAffect example

philippitts, 2023-06-30:
As mentioned by @wjcroft on #139:

> On your main page:
> 
> https://docs.openbci.com/Examples/ExamplesLanding/
> 
> You should definitely link to the MindAffect project, as it is a superior BCI paradigm to both MI and SSVEP. There is a long Forum thread with videos and links.
> 
> https://openbci.com/forum/index.php?p=/discussion/2783/mindaffect-announces-open-source-release-of-their-cvep-bci-speller-games-etc/
> 
> and
> 
> https://openbci.com/community/mindaffect-releases-open-source-bci-software/

wjcroft, 2023-06-30:
You don't actually need an OpenBCI written 'example' for MindAffect, as they already have abundant docs and tutorials online.

https://mindaffect-bci.readthedocs.io/en/latest/

You just need to point there from the Docs:

https://docs.openbci.com/Examples/ExamplesLanding/

Perhaps also point to the Forum and Community links (first post) as well.

wjcroft, 2023-06-30:
Motor Imagery BCI paradigm is decades behind what has been done with modern cVEP based system. 

https://www.youtube.com/watch?v=gk05sTk6dmI

wjcroft, 2023-09-20:
To reiterate, 'example' is not needed. Just link to the MindAffect docs, website, Forum pages, etc. MindAffect already has detailed tutorial info.

wjcroft, 2024-11-13:
See TWO recent OpenBCI Community tutorials on MindAffect.

https://openbci.com/community/mind-controlled-robot-openbci-mindaffectbci-maqueen-v2/

https://openbci.com/community/mind-controlled-game-openbci-mindaffectbci-unity/