OpenBCI/Documentation #182: Update Simblee Link in Ganglion Programming Tutorial

philippitts, 2023-07-26:
The [Ganglion Programming Tutorial ](https://docs.openbci.com/Ganglion/GanglionProgram/) references a broken Google Drive link that has a specific compiler and toolset for an old version of MacOS. This should be updated to direct users to simply use the "Additional Boards Mnager URLs" field of the Arduino preferences. The below link has been updated with a working JSON file.

```
https://openbci-simblee.s3.amazonaws.com/package_simblee_index.json
```

It would be best to host these files at prettier URLs. This could be done via the website, documentation or possibly in the repository itself. We should explore options. The JSON file should be updated if the Simblee tar.gz archive is moved to a different location.

wjcroft, 2023-07-26:
Related issue thread:

https://github.com/OpenBCI/OpenBCI_Ganglion_Library/issues/17

philippitts, 2023-07-27:
The package configuration file listed above now simplifies the process of setting up an environment to compile Ganglion firmware. See the related issue thread linked by @wjcroft for more details.

closed 2023-08-08 by philippitts