OpenBCI/Documentation #185: Document brainflow environment variable on MacOS

philippitts, 2023-08-07:
We need to update the documentation to notify developers on MacOS that they will need to set the following environment variable when running from Processing.

```
export ld_library_path=/path/to/libraries/brainflow
```

_Originally posted by @philippitts in https://github.com/OpenBCI/OpenBCI_GUI/issues/1175#issuecomment-1668486528_

retiutut, 2023-09-20:
Idk if this is still a thing. Needs to be confirmed using new GUI v6.

ranyabel, 2025-03-17:
This was tested and did not need to change the library path. Steps done: 
downloaded the processing SDK in downloads page in OpenBCI website. ran processing, opened GUI file from processing, GUI opened and functional.

closed 2025-03-17 by ranyabel