OpenBCI/Documentation #234: Add instructions for cyton reset

zoeis52, 2025-02-03:
Currently no instructions for how to reset cyton, or why you should or should not, or how to recover from an accidental reset.

retiutut, 2025-02-07:
This issue should involve collaboration and discussion from assignees before attempting to write documentation.

Very high priority as we get Zendesk requests when users have pressed the reset button and it could cause a hardware replacement in some cases depending on how exactly the buttons were pressed.

shirleyzhang867, 2025-03-14:
This issue can be resolved.

If a user taps the RST button and _no other buttons_ on the Cyton board, the board will work normally.

If the blue LED is blinking due to accidentally pressing buttons on the front of the Cyton board, functionality can be restored by:
1. make sure the battery is plugged into the Cyton board
2. set the board switch to 'PC'
3. tap the RST button once.
4. User will see that the LED stops blinking, and stays ON
5. Confirm board is functional by connecting the board to the OpenBCI GUI

tharun-OpenBCI, 2025-03-17:
Thanks @shirleyzhang867 for the breakdown! I can work on this by adding a section to the [Troubleshooting page](https://docs.openbci.com/Troubleshooting/TroubleshootingLanding/) and cover the following things with videos:

1. How to check if RST button was pressed and board is working normally
2. Tutorial on how to restore functionality if blue LED is blinking
3. When to contact OpenBCI because board is non-functional due to RST button presses

retiutut, 2025-03-26:
Brought up in a conversation with a group of new users.

retiutut, 2025-03-26:
>There is no reason to press the RST button, except to upload custom firmware. A rare scenario.

>If someone accidentally presses the buttons on the front of the Cyton board, they can tap the RST button to restore functionality.

tharun-OpenBCI, 2025-03-28:
The only way I was able to trigger a blinking LED on the Cyton board was by either following the Programming guide or after the board is on, by holding down the PROG button and then clicking the RST button. 

In either scenario, pressing the RST button or power cycling makes the board go back to a normal state. This can be documented in the tutorial to exist a "blinking blue LED" state.

PR #262 (2025-04-22): Add Information and Tutorial on What to do When Reset Button is Pressed
Closes #234
closed 2025-04-22 by retiutut